Cristov9000/RaspberryPi-HandHeld #5: Power supply doesn't turn off after switch is turned off

kvchen, 2016-08-17:
I'm running into an issue where the ATtiny won't power off the Adafruit 1000C power supply after the switch is turned off. When I power off the switch, the screen blanks and the Raspberry Pi goes through its usual shutdown routine, finishing with just the static red power light. After about 10 seconds, both the power supply light and the Raspberry Pi power lights are still on, indicating that the power supply was not switched off. Trying to turn the switch back on will turn the power light on (on the front case), but the Raspberry Pi will remain off. I end up having to disconnect the battery in order to shut everything down. Aside from that, everything works fine - when I connect the battery, everything is powered off until I flip the switch on; it's just shutting down that doesn't quite work.

My guess is that the ATtiny doesn't receive the message that the Raspberry Pi is shutting down, so it never ends up killing the power. I don't have the gerber for the `SGGBackBoard`, so it's pretty hard for me to debug this one.

First of all, when does `rpistate` get set to LOW? `pm4.c` shows that it's defined on `PB3`, but I'm not sure where that's hooked up to on the PCB. Secondly, the main circuitboard shows `RX`, but the associated GPIO pin is 18, not RX (which is GPIO 15).

I'm currently using a Raspberry Pi v3 and the included ATtiny85 that I got from the SuperGameGirl products page.

kvchen, 2016-08-27:
Resolved. The issue is that the TX pin on the RPi3 doesn't quite hit a low enough voltage when the Pi shuts down, so connect the TX pin on the SuperGameGirl board to pin 23 on the Raspberry Pi 3 instead. Nothing else needs to be changed.

closed 2016-08-27 by kvchen

christianhaitian, 2016-10-21:
Just a correction on this.  It should be connect the TX pin on the SuperGameGirl board to GPIO 23 (PIN 16) on the Raspberry PI 3 to resolve the shutdown issue for this project using the latest version of RetroPie.