Cristov9000/RaspberryPi-HandHeld #8: Instructions (detailed imgur) are no longer available

pyrocide, 2017-08-27:
I finally have all the parts required but the original imgur post is gone, archive does not include all photos making replication much more difficult.

BioHaZard1, 2017-08-27:
This will help you out https://imgur.com/a/KuBRE#eaF57E6

I also have a complete copy of the original guide too. Not sure why the project is being rebranded and the guide got taken down. Strange.

pyrocide, 2017-08-27:
Ah much appreciated. I started the desoldering hell and holy shit it sucks.

BioHaZard1, 2017-08-28:
Yeah, I didn't find it too hard, look me about 30-40 minutes to remove all components. I used a solder sucker, it made the process much easier.

mattc74, 2017-10-04:
You can view the original guide via. the wayback machine, here:

https://web.archive.org/web/20160105060629/https://imgur.com/a/gFhk7

jrdnlc, 2017-10-04:
@BioHaZard1 You mind sharing your copy of the conplete guide?

BioHaZard1, 2017-10-07:
@pyrocide 
@jrdnlc 

Re-Uploaded guide:
https://imgur.com/a/JKNMZ

jrdnlc, 2017-10-07:
Thank you @BioHazard1