Cristov9000/RaspberryPi-HandHeld #9: Pi shut down but won't restart after using the spdt switch

Edjuh82, 2017-09-27:
When i turned off the pi using the spdt switch it won't restart after turning the switch back on. My powerboost also stays on during the shut off. It turns off the screen and turns it on but not the pi. 

Anyone know how to solve this?

Using a raspberry pi 3 and used this guide aswell as the other one posted on reddit.

mattc74, 2017-10-04:
The pi3 is different in that it uses a software UART. The physical uart is used for bluetooth on the pi3 so it's now a software emulated uart on the gpio pins. 

I had to switch on the console (opposite to pi2) in raspi-config to generate a voltage for the attiny to sense, I also dropped a small capacitor (10uF) across to smooth it out, works perfectly!

spedga, 2018-02-13:
@mattc74 I'm also having this issue. Would you mind explaining "I had to switch on the console (opposite to pi2) in raspi-config to generate a voltage for the attiny to sense" in a bit more detail? I'm fairly new to editing configs with the raspberry pi. It seems like this is the last step I have before screwing everything in place and finally finishing this project and getting down to some gaming!

mattc74, 2018-02-13:
You can either:

Enable the console in sudo raspi-config

Or

 you need to change enable_uart=1 in /boot/config.txt

spedga, 2018-02-14:
I did both of these @mattc74 but it's still not booting after shutting down. Do you have anymore ideas for troubleshooting steps I could try?

BioHaZard1, 2018-03-18:
Must be a problem with wiring then as I have just done this successfully.